Ryzee119/ogx360_t4 #8: usb64 like PCB

SavageCore, 2021-02-01:
Hey

Just wondering if you were considering creating a usb64 like board for this? One teensy 4 controllers?

rizaumami, 2021-02-02:
One teensy for four controllers would be amazing.

Ryzee119, 2021-02-02:
That would be very awesome. 

Emulating multiple usb devices at once is very tricky and OG xbox doesnt seem to accept composite USB devices which would make it really easy.

Teensy4.1 and almost all microcontrollers only have one to two usb controller built in.

I have found a library to bitbang usb protocol which is very havky but the Teensy may have to power to bitbang multiple at once. 

Once i work out i good method for this ill make a nice breakout board.

You can get really low cost cortex m0 controllers with usb controllers built in for like $1. This may be the easiest bey

Aasikki, 2022-08-15:
Even two controllers per teensy would be amazing :)
These things cost quite a bit (almost 40€ where I live), so that would be a significant cost saving as well.