Ryzee119/ogx360_t4 #13: Simultaneous Duke and SDcard support

Verasl, 2021-08-29:
Hey,

I don’t know if I’m building this incorrectly or missing a step but being able to have both duke emulation and the sdcard xmu emulation in a single module would be amazing.
Or if processing power is a concern, being able to build all emulation functionality but somehow switch the modes once flashed would be huge.
Working on a few of these for an internal mod but having to re-flash to get the different modes make internal installation a bit of a pain.

Cheers!

Ryzee119, 2021-08-29:
Yes its definitely possible. The way ive written the drivers makes it a bit hard atm.
I have a branch called composite which aimed to solve this. (I.e you could have 3 controllers, 4 xmus, 1 steel battalion and a headset at all once!)
Although the xbox doesnt support this anyway, but xmu+controller is definitely supported.

Ill have to revisit that branch and see where i left off. I was having problems with the xmu stuff from memory

Verasl, 2021-08-30:
Awesome thanks! Either way this is great stuff, I had the ogx360, but like the more simple setup this had, plus didn’t have to think about hooking a usb hub up for powering 4 controllers.
Keep up the epic contributions!

Ryzee119, 2022-02-01:
This should be possible now.

Configure platformio.ini and set these to one then compile.
    -DXID_DUKE=1
    -DMSC_XMU=1

closed 2022-02-01 by Ryzee119