Ryzee119/ogx360_t4 #14: Forza Horizon 5 Controller issues

splinterCellFan, 2021-12-17:
Hi,
First let me thank you for  your work. Having a blast with it.
So i got some issues with this controller: https://www.xbox.com/en-US/accessories/controllers/forza-horizon-5-limited-edition.

If i plug it in and then start the xbox, the first time it doesnt work. Inputs  are not reflecting on the screen. If i unplug and then plug it again(xbox controller) then it works, for some time.
I'm testing it by playing Forza Motorsport 1 and after 20-30 seconds the controller stops responding(seems to be stuck on the last received commands ) and then it disconnects . Then it reconnects by itself.  
So something is not working right for this controller or there is an issue with series x controller type in the code as the other(xbox one and 360 controllers are working fine, tested them for hours)

If i switch between controllers at one point none will work so i have to restart the xbox/cut power from the board.

Ryzee119, 2022-02-01:
Please try latest build and let me know if this is still present

splinterCellFan, 2022-02-13:
Hi, took latest last night. 
Bad news, now the forza controller(series x) doesnt work at all and the and the one/one x, does exactly like the series x did.
By any chance do still have the hex around, the one that you tested the first time?

splinterCellFan, 2022-06-12:
Today as of 06.12.2022,after i updated the firmware of the xbox controller to the latest i was able to play without any issues. 
So the issue was something in the firmware.  No surprise here, Microsoft!
I tihnk this one can be closed now.

closed 2022-06-12 by splinterCellFan