Ryzee119/ogx360_t4 #19: [Feature Request] Xbox Wireless Adapter Support

GoTeamScotch, 2022-05-26:
_Continuing the conversation from [ticket #6](https://github.com/Ryzee119/ogx360/issues/6) on the original Ogx360 repo._

Request: add support for [Xbox Wireless Adapter](https://www.xbox.com/en-US/accessories/adapters/wireless-adapter-windows). Hopefully the extra power the Teensy and its ARM Cortex-M7 makes this possible now.

[image]

The adapter uses a MediaTek MT7600U chip which handles 802.11b/g/n transmission (used by Xbox wireless controllers). There is another open source project, [medusalix/xone](https://github.com/medusalix/xone), that might offer some inspiration for how to interact with the Wireless Adapter. This project is a Linux kernel driver for the 1790 adapter and its MT76 chip. The heavy lift of reverse-engineering the adapter's protocol [has been done](https://github.com/medusalix/xone/blob/master/transport/mt76_defs.h) and could be referenced for faster development here.

Support this adapter would also get around [8bitdo's incompatibility](https://www.8bitdo.com/wireless-usb-adapter/) with Xbox wireless controllers that have been updated to the new (Sept 2021) firmware, which switches from bluetooth _classic_ to _low-energy_ mode.

This adapter supports multiple controllers connected at the same time. Currently, ogx360_t4 emulates one Xbox gamepad at a time. If that ever changes, this adapter could be a convenient (and cheaper) alternative to using multiple 8bitdo's + a USB hub.

_Note: there are two models for the wireless receiver. The newer/smaller "1790" model (Jan 2018) and the older/larger "1713" model (Oct 2015)._

DrunkenJediMaster, 2023-03-25:
Adding my support for this idea

burritobike, 2024-09-25:
I mislead myself into believing this would work with the larger xbox one wireless adapter before reading this. I think it would be awesome to have these dongles supported.