Ryzee119/ogx360_t4 #22: Disconnects

Dlang6487, 2022-12-11:
While playing Burnout Revenge with an Xbox Series controller, it will stop responding. I have to unplug the adapter from the Xbox and plug it back in. It does not say controller unplugged in game, it just stops responding to buttons presses while still showing that it is connected. This happens after 30 mins or so of gameplay sporadically.

RobinNorling, 2023-01-30:
> While playing Burnout Revenge with an Xbox Series controller, it will stop responding. I have to unplug the adapter from the Xbox and plug it back in. It does not say controller unplugged in game, [...]

Had this happen to me a couple of times as well while playing Tony Hawk's Underground, although only 2 or 3 times.
As far as i can tell it's getting stuck somewhere inside the `tud_task` call inside the main loop but that's as far as i've been able to check, from my limited testing it does however seem like the call is never returned.
Whether it could be as simple as to setting a more sensible timeout value or if it's a deeper issue i have to dig deeper into.

I'm new to this myself but i'll look into it once i have some spare time. As a side note, i messed up when wiring it up so it is connected to port 2, whether that has something to do with it i couldn't tell you at this point but the `tud_task_ext` call timeout in usbd.h on line 55 is set to virtually infinite so my best guess is that the queue receive call (usbd.c:483) doesn't finish.

I should mention that this is mostly speculation at this point and i haven't thoroughly tested anything.

RobinNorling, 2023-02-06:
After some testing it seems like updating to the latest version of tinyusb was all it took to fix my issue.
@Dlang6487 give this a shot, can't guarantee that it will solve the issue but it's a good place to start

Ryzee119, 2023-02-10:
Tinyusb just released v15 so will update submodule soon

ledah90, 2023-02-17:
I'm having the same issue. I believe I updated tinyusb correctly and the issue is still present. For reference, I am using a xbox one controller on latest firmware and 8bitdo adapter v2 on 1.03 firmware. I will try compiling again when the submodule is updated to verify I didn't get something correct while updating.

RobinNorling, 2023-02-17:
> I'm having the same issue. I believe I updated tinyusb correctly and the issue is still present. For reference, I am using a xbox one controller on latest firmware and 8bitdo adapter v2 on 1.03 firmware. I will try compiling again when the submodule is updated to verify I didn't get something correct while updating.
> 
> Edit: I forgot to mention I have only tested the controller on Halo Combat Evolved.

How often does the issue occur? I tried playing Halo Combat Evolved for a little over an hour using an 8bitdo v2 (firmware 1.00 though), and xbox one controller and had no issues.

ledah90, 2023-02-17:
Pretty frequent. Maybe 5 or 10 mins of play time. I tried it on Doom Resurrection as well and the same thing occurs, but takes longer to happen. I can try to downgrade my adapter to 1.00 firmware and test again. When it happens, I am still getting vibration feedback from being hit or damaged in game, but the controls stop working.

s3anami, 2023-02-27:
i have the same issue btw going to try the firmware change on the 8bitdo

cmcginty, 2023-05-27:
I'm still seeing disconnections. So far I rebuild the latest firmware and re-flashed, and downgraded the 8bitdo v2 adapter to 1.0. Anything else to try at this point?