Ryzee119/ogx360_t4 #23: build problem

djsteverobinson, 2022-12-20:
hi i Clone this repo recursively build with PlatformIO i just get fatal error: any help thanks 

Warning! `src_filter` configuration option in section [env:XMU] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:DUKE] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:STEELBATTALION] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Processing XMU (platform: teensy; board: teensy41; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/teensy/teensy41.html
PLATFORM: Teensy (4.17.0) > Teensy 4.1
HARDWARE: IMXRT1062 600MHz, 512KB RAM, 7.75MB Flash
DEBUG: Current (jlink) External (jlink)
PACKAGES: 
 - framework-arduinoteensy @ 1.157.220801 (1.57) 
 - tool-teensy @ 1.157.0 (1.57) 
 - toolchain-gccarmnoneeabi @ 1.50401.190816 (5.4.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 91 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/XMU/src/duke.cpp.o
Compiling .pio/build/XMU/src/main.cpp.o
Compiling .pio/build/XMU/src/steelbattalion.cpp.o
Compiling .pio/build/XMU/src/usbd/tusb_print_helper.c.o
src/usbd/tusb_print_helper.c:1:20: fatal error: printf.h: No such file or directory

****************************************************************
* Looking for printf.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:printf.h"
* Web  > https://registry.platformio.org/search?q=header:printf.h
*
****************************************************************

 #include "printf.h"
                    ^
compilation terminated.
*** [.pio/build/XMU/src/usbd/tusb_print_helper.c.o] Error 1
src/steelbattalion.cpp:2:18: fatal error: tusb.h: No such file or directory

**************************************************************
* Looking for tusb.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:tusb.h"
* Web  > https://registry.platformio.org/search?q=header:tusb.h
*
**************************************************************

 #include "tusb.h"
                  ^
compilation terminated.
src/duke.cpp:2:18: fatal error: tusb.h: No such file or directory

**************************************************************
* Looking for tusb.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:tusb.h"
* Web  > https://registry.platformio.org/search?q=header:tusb.h
*
**************************************************************

 #include "tusb.h"
                  ^
compilation terminated.
*** [.pio/build/XMU/src/steelbattalion.cpp.o] Error 1
*** [.pio/build/XMU/src/duke.cpp.o] Error 1
src/main.cpp:2:25: fatal error: USBHost_t36.h: No such file or directory

[... 156 lines cut ...]
src/steelbattalion.cpp:2:18: fatal error: tusb.h: No such file or directory

**************************************************************
* Looking for tusb.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:tusb.h"
* Web  > https://registry.platformio.org/search?q=header:tusb.h
*
**************************************************************

 #include "tusb.h"
                  ^
compilation terminated.
*** [.pio/build/STEELBATTALION/src/steelbattalion.cpp.o] Error 1
========================================================================== [FAILED] Took 0.90 seconds ==========================================================================

Environment     Status    Duration
--------------  --------  ------------
XMU             FAILED    00:00:00.972
DUKE            FAILED    00:00:00.909
STEELBATTALION  FAILED    00:00:00.900
===================================================================== 3 failed, 0 succeeded in 00:00:02.781 =====================================================================

 *  The terminal process "platformio 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.

Ryzee119, 2022-12-20:
You definitely have not cloned to repo properly.

`git clone https://github.com/Ryzee119/ogx360_t4.git --recursive`

closed 2022-12-20 by Ryzee119

djsteverobinson, 2022-12-21:
hi i go it to work i left Visual Studio Code in the downlod folder is now working when i put it in the applications fille programing  worked
new problem is the xbox one controler sinks rumbels but nothing moves ? i have 3 8bits 2 red one black tride every fw on them same problem  i have enable Xinput mode i see the light blink the soldering is good any advice thanks  
 

*  Executing task: platformio run --target upload 
Processing DUKE (platform: teensy; board: teensy41; framework: arduino)
---------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/teensy/teensy41.html
PLATFORM: Teensy (4.17.0) > Teensy 4.1
HARDWARE: IMXRT1062 600MHz, 512KB RAM, 7.75MB Flash
DEBUG: Current (jlink) External (jlink)
PACKAGES: 
 - framework-arduinoteensy @ 1.157.220801 (1.57) 
 - tool-teensy @ 1.157.0 (1.57) 
 - toolchain-gccarmnoneeabi @ 1.50401.190816 (5.4.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 91 compatible libraries
Scanning dependencies...
Dependency Graph
|-- SD @ 2.0.0
|   |-- SdFat @ 2.1.2
|   |   |-- SPI @ 1.0
Building in release mode
Linking .pio/build/DUKE/firmware.elf
Checking size .pio/build/DUKE/firmware.elf
Building .pio/build/DUKE/firmware.hex
Configuring upload protocol...
AVAILABLE: jlink, teensy-cli, teensy-gui
CURRENT: upload_protocol = teensy-gui
Uploading .pio/build/DUKE/firmware.hex
Opening Teensy Loader...
================= [SUCCESS] Took 3.44 seconds =================

Processing TEST (platform: teensy; board: teensy41; framework: arduino)
---------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/teensy/teensy41.html
PLATFORM: Teensy (4.17.0) > Teensy 4.1
HARDWARE: IMXRT1062 600MHz, 512KB RAM, 7.75MB Flash
DEBUG: Current (jlink) External (jlink)
PACKAGES: 
 - framework-arduinoteensy @ 1.157.220801 (1.57) 
 - tool-teensy @ 1.157.0 (1.57) 
 - toolchain-gccarmnoneeabi @ 1.50401.190816 (5.4.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 91 compatible libraries
Scanning dependencies...
Dependency Graph
|-- SD @ 2.0.0
|   |-- SdFat @ 2.1.2
|   |   |-- SPI @ 1.0
Building in release mode
Linking .pio/build/TEST/firmware.elf
Checking size .pio/build/TEST/firmware.elf
Building .pio/build/TEST/firmware.hex
Configuring upload protocol...
AVAILABLE: jlink, teensy-cli, teensy-gui
CURRENT: upload_protocol = teensy-gui
Uploading .pio/build/TEST/firmware.hex
Unable to open [no_device] for reboot request
================= [SUCCESS] Took 8.46 seconds =================

Environment    Status    Duration
-------------  --------  ------------
DUKE           SUCCESS   00:00:03.441
TEST           SUCCESS   00:00:08.456
================= 2 succeeded in 00:00:11.897 =================
 *  Terminal will be reused by tasks, press any key to close it.

Ryzee119, 2022-12-21:
I think you're programming the test firmware. change the environment to DUKE.
You change the environment on the PIO toolbar as the bottom of the vscode window

djsteverobinson, 2022-12-21:
that did the job thanks for the helps