Ryzee119/ogx360_t4 #26: Micro SD card support

ledah90, 2023-03-16:
I have tested 4 different brand micro sd cards all formatted using the SD Card Formatter suggested in the instructions. I am unable to do anything on any card I place into the device. It is never registered as an available device. Does it need to be some special cluster size or brand? The brands I have tried are Sandisk, Samsung, Monster, and PNY. All micro SD cards are 4gb or less and FAT32 format with default cluster size.

Ryzee119, 2023-03-17:
Hmm that's interesting

Have you enabled `-DMSC_XMU=0` here https://github.com/Ryzee119/ogx360_t4/blob/be37b41775778137f6e1918245771dc31b6afea9/platformio.ini#L62 when compiling?

Also try increasing this a bit. The max is 4096. https://github.com/Ryzee119/ogx360_t4/blob/be37b41775778137f6e1918245771dc31b6afea9/platformio.ini#L72

ledah90, 2023-03-17:
Yes, I have DMSC set to 1 and size set to 2048. I cleaned the solution and rebuilt again and I get "The memory unit you inserted isn't functioning; it may be damage." This is the same result on all 4 different cards. Does it need to match 4098 if I have a 4GB card?

Ryzee119, 2023-03-17:
Nah that size doesn't have to match the SD card size. 
Looks like you're doing everything right. it might be a bug. I'll have to look into it

Thanks for reporting the issue

Ryzee119, 2023-03-17:
Does it work in your PC?

It should on first power up read the SD card for XMU_MSC.bin. if not present it will create it.

If you plug it into your PC does it show up as a mass storage device with that file?

ledah90, 2023-03-21:
If I place the SD card in my computer, there is a file called XMU_MSC.bin that is 0 bytes. If I change the DXMU_SIZE_MB to 8mb, it creates the file as it should. With this 8mb file, I am still presented with the same "The memory unit you inserted isn't functioning; it may be damage." message. I have tried this on a 1GB, 2GB, and 4GB micro SD card.

vwbug2007, 2023-04-09:
have anyone have issue with 3d printed case usb port crushing the sd card due to clearance issue i had one broken microsd card because of that i think the stl file needs to be changed to allow more clearnce ....if you have issues with microsd card check if its not bent or cracked in half....

vwbug2007, 2023-04-09:
[image]
[image]
 See?

Ryzee119, 2023-04-09:
@vwbug2007 this repo has no 3d printed case so we can't help here.

Please keep this issue clear as it relates to SD card firmware issues

vwbug2007, 2023-04-09:
> Yes, I have DMSC set to 1 and size set to 2048. I cleaned the solution and rebuilt again and I get "The memory unit you inserted isn't functioning; it may be damage." This is the same result on all 4 different cards. Does it need to match 4096 if I have a 4GB card?

same error as your hmmm....

vwbug2007, 2023-04-09:
> If I place the SD card in my computer, there is a file called XMU_MSC.bin that is 0 bytes. If I change the DXMU_SIZE_MB to 8mb, it creates the file as it should. With this 8mb file, I am still [...]

yeah samething i can see the file only obyte when i had it compiled with size 4096 on an old 8gb card ...

vwbug2007, 2023-04-09:
one more thing on formatting microsd card to fat32 at what cluster size?

ledah90, 2023-07-29:
Ok, I tested a few things out and figured out quite a bit. First the micro SD card I tested with is a SanDisk 2GB and formatted at 16byte cluster size. All tests were done on a 1.6 Xbox console using one 8bitdo v2 adapter and Xbox one controllers via Bluetooth. I tested viewing the memory card on a few dashboards. UnleashX and EvoX do not recognize the micro SD bin as a memory card or device. UIX Lite does see there is a memory card, but registers it as corrupted and unusable. The MS dashboard however sees the memory card just fine and can read and write to it. I also tested using a disc based game and the same game installed internally, but neither can see the memory card. You can use a USB hub connected to the controller USB port to connect 2 USB drives to act as a memory cards as well as use the ogx360. Hopefully some of this information is helpful in giving you some kind of direction. What console version and dashboard were you testing your original code on? I have several different consoles I can test against and want to replicate what you are using if possible.

djholt, 2024-08-04:
@ledah90 and anyone else who might be experiencing SD card read/write failures: my initial testing of the XMU emulation showed a lot of inconsistencies and failures like you're describing. My initial logic was to use older/smaller SD cards, because the Xbox is picky with memory storage, requires 4GB or less, etc. But the opposite strategy tends to work better here: try using newer/faster/larger SD cards.

The Teensy itself is mounting and accessing the SD card, not the Xbox directly, so it seems the better performance you can get from the SD card itself, the better the emulation works, at least in my testing. Larger SD cards generally do have better read/write performance, so remember that while the max XMU size is 4096MB (just under to be safe) due to filesystem limitations, the SD card itself can be larger. I'm using a Samsung EVO 128GB micro SD card with a 4095MB XMU image written to it. It's working great for me.

I hope this helps someone seeing intermittent performance from the XMU emulation.

Also, for anyone interested in mounting the XMU image for reading/writing on a PC, [refer to this](https://github.com/mborgerson/fatx/issues/57#issuecomment-2267227670).

Ryzee119, 2024-08-04:
Right, im fairly sure this is to do with SD card latency. The timeouts on xbox are very strict and all the overhead talking with the SD card can exceed the timeout sometimes and cause the Xbox to bail out. 

Maybe some clever use of buffering into RAM could help here.

The way the code works is that is presents itself to the xbox as whatever the file of  `XMU_MSC.bin` on the SD card os so the size of the SD card does not matter.

If you get zero bytes on the file, try deleting the file on the SD card or generating your own `XMU_MSC.bin` file that is 8MB (can just contain zeros or whatever). This is a bug that doesnt recovering very well.

djholt, 2024-08-04:
Makes a lot of sense. I do find it a little ironic that SD card latency becomes an issue over USB 1.1 bus speeds. Any recently-manufactured SD card really should be able to keep up, but not always the case.

And yes, I did observe more stability with slower cards when pre-initializing the .bin file to its intended size. I went a step further and wrote the fatx header to the file at the same time, but that was mostly after seeing some strange behavior from the Xbox when formatting XMUs >8MB.

Buffering in RAM is a great idea, but honestly, it's working really well for me currently with not-ancient SD cards.

Thanks @Ryzee119 !